moveit/moveit_pr2 #58: Adding/Removing Objects, Dual-arm pose goals missing in move_group_interface_tutorial.py

rahul003, 2015-09-24:
Tutorials for Adding/Removing Objects, and Dual-arm pose goals are incomplete/missing in moveit_pr2/pr2_moveit_tutorials/planning/scripts/move_group_interface_tutorial.py .

The same tasks are shown in the cpp tutorial

sweetsaikiran690, 2017-08-07:
For me also, Same problem.

Is it possible to provide the Sample code for that.

davetcoleman, 2017-08-17:
Are you aware the PR2 tutorials have been moved to https://github.com/ros-planning/moveit_tutorials/?

sweetsaikiran690, 2017-08-17:
Sorry. I am not aware of this.

But in above link also, I didn't find move_group_interface_tutorial.py file

davetcoleman, 2017-08-17:
Try https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/pr2_tutorials/planning/scripts/move_group_python_interface_tutorial.py

closed 2017-08-17 by rahul003

sweetsaikiran690, 2017-08-18:
Yes. But in that I am unable to find code for this concept

sweetsaikiran690, 2017-08-18:
## Adding/Removing Objects and Attaching/Detaching Objects