moveit/moveit_pr2 #64: pr2_moveit_tutorials is missing a dependency on shape_tools

tfoote, 2016-02-05:
It's the one regression in indigo at the moment. I suspect it was relying on a transitive dependency which was cleaned up so now it's broken. 

This is blocking the build of moveit_pr2 and moveit_full_pr2

http://build.ros.org/view/Ibin_uS64/job/Ibin_uS64__pr2_moveit_tutorials__ubuntu_saucy_amd64__binary/10/console

```
CMakeFiles/pick_place_tutorial.dir/src/pick_place_tutorial.cpp.o -c /tmp/binarydeb/ros-indigo-pr2-moveit-tutorials-0.6.1/pick_place/src/pick_place_tutorial.cpp
00:12:36.857 /tmp/binarydeb/ros-indigo-pr2-moveit-tutorials-0.6.1/pick_place/src/pick_place_tutorial.cpp:42:46: fatal error: shape_tools/solid_primitive_dims.h: No such file or directory
00:12:36.857  #include <shape_tools/solid_primitive_dims.h>
00:12:36.857                     
```

IanTheEngineer, 2016-02-05:
Thanks for the report, Tully. I actually resolved this the other day:
https://github.com/ros-planning/moveit_pr2/pull/63
and cut a new release this morning:
https://github.com/ros-planning/moveit_pr2/releases/tag/0.6.2
Unfortunately, I seem to be lacking write permissions required to push to ros-gbp org (ros_planning team) for bloom.

IanTheEngineer, 2016-02-05:
One thing to note, we seem to have a version mismatch on moveit_commander and moveit_plugins between the most recent releases and those that were merged into rosdistro. @130s noticed this:

https://github.com/ros/rosdistro/pull/10378#issuecomment-177343813
moveit_commander: rosdistro says 0.5.7-1, sand should be 0.6.0-0
https://github.com/ros/rosdistro/pull/10374#issuecomment-177344429
moveit_plugins: rosdistro says 0.5.6-2, sand should be 0.5.7-0

We shouldn't do a sync until this discrepancy is rectified.

tfoote, 2016-02-05:
@IanTheEngineer I see you on the team now. Can you rerun bloom?

IanTheEngineer, 2016-02-05:
I can re-run it tonight on my home workstation. My work machine's environment is non-standard, as you noticed: http://answers.ros.org/question/223663/dependencies-not-satisfied-ros-indigo-gazebo5-msgs/

If anyone else in @ros-planning wants to rebloom those 3 packages in the meantime, I would not object

mikeferguson, 2016-02-05:
I can also help out --- after 4pm today

IanTheEngineer, 2016-02-06:
Resolved :) https://github.com/ros/rosdistro/pull/10433

closed 2016-02-06 by IanTheEngineer

cdrwolfe, 2017-02-26:
Anyone know what the actual fix was for this? Seems to be replicated on the turtlebot_arm package as well when building in catkin:
```
fatal error: shape_tools/solid_primitive_dims.h: No such file or directory
 #include <shape_tools/solid_primitive_dims.h>
```
https://github.com/corb555/turtlebot_arm

130s, 2017-02-26:
@cdrwolfe https://github.com/ros-planning/moveit_pr2/pull/63?