moveit/moveit_pr2 #68: Gripper Issue

ilaplace, 2016-06-17:
Hi there;
I wasn't able to open the gripper of pr2 with your code.  I tried this stuff but it didn't work either. Do you have any idea what should i do to open the gripper with moveit? 
Thank you in advance

` g.pre_grasp_posture.joint_names.resize(1, "r_gripper_joint");
    g.pre_grasp_posture.points.resize(1);
    g.pre_grasp_posture.points[0].positions.resize(1);
    g.pre_grasp_posture.points[0].positions[0] = 1;
    g.pre_grasp_posture.points[0].effort.resize(1);
    g.pre_grasp_posture.points[0].effort[0]=1;`

v4hn, 2016-06-19:
Hey @ilaplace, I'm sorry, but this is way to unspecific to help you.
Could you describe what exactly you are trying to do and provide the full code that fails to work?

ilaplace, 2016-06-20:
First of all, thank you for your respond;
I used exactly the same code in the " moveit_pr2/pr2_moveit_tutorials/pick_place/src/pick_place_tutorial.cpp" tutorial. I thought while the robot is moving it supposed be open it's grip. But it doesn't happen, In Rviz it seems to attach with the object without opening  its grip. I tired to add 
`g.pre_grasp_posture.points[0].effort[0]=1;``
code but it didn't work either.

dornhege, 2016-07-26:
Did you have a PR2 (or simulated) in the background or just RViz. I believe the problem is that the PR2 gripper joints that actuate the gripper are different from the actuated ones, i.e., you specify a joint that will make the (real) PR2 gripper move (like an internal drive motor), which in turn actuators parts of the hand that are actually visible.
Thus, to move an visualize the posture needs to be extended. In another module I have the following posture, which you can test:

    joints : ['r_gripper_motor_screw_joint',
        'r_gripper_l_finger_joint',
        'r_gripper_r_finger_joint',
        'r_gripper_r_finger_tip_joint',
        'r_gripper_l_finger_tip_joint']
    pregrasp_posture : [1.0, 0.477, 0.477, 0.477, 0.477]

Leaving this open as the tutorial might need to be amended.

v4hn, 2016-07-28:
So the bug is that our fake controllers do not (and in practice *can't*)
support controlling joints, but require all joints set?
That makes sense.

I just added complete open/close states in https://github.com/v4hn/moveit_pr2/tree/pick_place_demo_with_gripper_motion ,
but I still do not see the gripper open/close. @dornhege Could you please have a look at the last commit there?
I don't see another reason why this shouldn't work at the moment.

To test, run `roslaunch pr2_moveit_config demo.launch` and `rosrun pr2_moveit_tutorials pick_place_tutorial`.

dornhege, 2016-07-28:
I'm not even sure if this is modeled in the URDF, so that we could handle this. It might also be just a joint that the PR2 received and then moves accordingly. On a real robot (or gazebo) this is never a problem. In MoveIt! only, this is. I can try, if your fix works.

However, I am aware of a (I assume) bug in the pick/place pipeline with the PR2, when doing reasoning on the PlanningScene (i.e. not executing) that needs to be fixed. This might be connected. I'll check this out, but might hold the fix until after the moveit merge.

v4hn, 2016-07-28:
No, this is not modeled in the URDF and I believe there's no need for us to have support for that in the fake controllers.
You can always specify the full state if you want to use your code with the fake controllers.

@dornhege: @Jntzko just found a bug in the code I pointed you to and I just added a pull request that fixes this issue.
Please review & merge.

dornhege, 2016-07-28:
I agree, we can't know anyways what other joints are activated how.

Looks good to me. I can see the gripper opening now!

PR #77 (2016-07-29): Improve Pick Place demo
Thanks to @ilaplace for pointing this out!

Fixes #68
closed 2016-07-29 by dornhege