moveit/moveit_pr2 #71: Release into Jade

130s, 2016-06-24:
Required by https://github.com/ros-planning/moveit_metapackages/issues/6

Currently blocked at https://github.com/PR2/pr2_kinematics/issues/6, which is also confirmed as follows:
```
$ wget https://raw.githubusercontent.com/ros/rosdistro/master/scripts/check_blocking_repos.py && chmod 755 check_blocking_repos.py
$ python ./check_blocking_repos.py --rosdistro jade --repositories moveit_pr2 --depth 1
The following repos cannot be released because of unreleased dependencies:
        moveit_pr2:
                pr2_kinematics
```

OwlCoder, 2016-07-22:
is this repo supposed to work with indigo?

130s, 2016-07-22:
Even binary is available for Indigo http://repositories.ros.org/status_page/ros_indigo_default.html?q=moveit_pr2 so yes.

130s, 2017-01-30:
Jade is approaching EOL in a few months. I suggest to drop this package support, and rather want to focus on Kinetic release. Feel free to reopen if there any objection.

closed 2017-01-30 by 130s