moveit/moveit_pr2 #72: Rviz cannot display robot properly

LongfeiProjects, 2016-07-08:
Hello, 

I had a problem regarding Rviz when launching most of launch files provided in this package. For example, 
$ roslaunch pr2_moveit_tutorials interactivity_tutorial.launch
The command above will give many repeated error messages regarding to URDF, such as:
[ERROR] [1468007825.024644293]: Link 'head_mount_link' is not known to URDF. Cannot disable collisons.
[ERROR] [1468007825.024656721]: Link 'head_mount_prosilica_link' is not known to URDF. Cannot disable collisons.
[ERROR] [1468007825.307693203]: Link 'head_mount_prosilica_link' is not known to URDF. Cannot disable collisons.
[ERROR] [1468007825.307758054]: Link 'head_mount_kinect_ir_link' is not known to URDF. Cannot disable collisons.
[ERROR] [1468007825.307775120]: Link 'head_mount_kinect_rgb_link' is not known to URDF. Cannot disable collisons.
[ERROR] [1468007825.307810335]: Link 'head_mount_link' is not known to URDF. Cannot disable collisons.
[ERROR] [1468007825.309565557]: Link 'head_mount_prosilica_link' is not known to URDF. Cannot disable collisons.
...

I ignored them at current stage since they are about collisions. These error messages shown up even when I launch "pr2_moveit_tutorial" came along with full ROS installation. 

Some additional messages are listed below:

TIFFFetchNormalTag: Warning, Incompatible type for "RichTIFFIPTC"; tag ignored.

[ INFO] [1468007240.357823223]: Loading robot model 'pr2'...
[ INFO] [1468007240.393144094]: #####################################################
[ INFO] [1468007240.393176446]: RVIZ SETUP
[ INFO] [1468007240.393185744]: ----------
[ INFO] [1468007240.393212335]:   Global options:
[ INFO] [1468007240.393222553]:     FixedFrame = /base_footprint
[ INFO] [1468007240.393254685]:   Add a RobotState display:
[ INFO] [1468007240.393264359]:     RobotDescription = robot_description
[ INFO] [1468007240.393273815]:     RobotStateTopic  = interactive_robot_state
[ INFO] [1468007240.393283255]:   Add a Marker display:
[ INFO] [1468007240.393293500]:     MarkerTopic = interactive_robot_markers
[ INFO] [1468007240.393302738]:   Add an InteractiveMarker display:
[ INFO] [1468007240.393312159]:     UpdateTopic = interactive_robot_imarkers/update
[ INFO] [1468007240.393337821]:   Add a MarkerArray display:
[ INFO] [1468007240.393364219]:     MarkerTopic = interactive_robot_marray
[ INFO] [1468007240.393377627]: #####################################################
[ INFO] [1468007240.394225757]: Not colliding
TIFFFetchNormalTag: Warning, Incompatible type for "RichTIFFIPTC"; tag ignored.

[ INFO] [1468007826.403953147]: Robot position: p(  0.585,  -0.188,   1.250) q(  0.000,  -0.888,   0.000,   0.460)

However, on the side of Rviz, the frame "base_footprint" looks like it does not exist:
Global Status/Fixed Frame No tf data.  Actual error: Fixed Frame [base_footprint] does not exist
Except "Add a RobotState display" gives a "dead" PR2 robot, nothing happens in Rviz.

I am running Ubuntu 14.04, ROS indigo. The code is cloned to "~/catkin/src/", and I was trying under "indigo-devel" branch. However, the "pr2_moveit_tutorial" coming along with ROS installation works fine even with "collision error message":
$ roslaunch pr2_moveit_tutorials move_group_interface_tutorial.launch

Can someone help me out with this problem. I would be glad to provide extra information if needed.

dornhege, 2016-07-26:
The robot displays fine for me (even with the error messages) and also TF is working. It seems like something is seriously wrong with the setup. Can you tell, what exactly you did, i.e., which parts you build from source? Best with your rosinstall file.

You can `export KINECT1=true` to get rid of the errors. This is due to the pr2_description being updated to include a Kinect. As this possibly needs to be fixed in all tutorials, I'll leave this open for now.

dk683, 2016-08-16:
Hello. I am experiencing the same problem.
Were you able to find the soltuion @LongfeiProjects?
Thank you in advance!

LongfeiProjects, 2016-08-17:
@dk683 I found a way to clear the error messages: Link "" is not known to URDF. Cannot disable collisons.

Please check here, and you there's another solution as well. [https://gist.github.com/davetcoleman/4256083](url)

dk683, 2016-08-17:
Aha! Thank you for your suggestion!

LongfeiProjects, 2016-08-17:
@dk683 There seems a problem in directly opening a link from comment. You can copy the link to a new window, then it should be all fine. :-)

dk683, 2016-08-17:
I got it! Thank you @LongfeiProjects!

LongfeiProjects, 2016-08-17:
@dornhege  Though I say this a bit late, but your comment is really appreciated. If I could have your comment a bit earlier, it will definitely save me much time on probing this issue. 

@dk683 I am very happy that this thread helps you as well.

Now the link error message can be considered as "solved", and I would like to close this thread.

The "RichTIFFIPTC" warning is an image related issue, it should has no influence in robot control. Some one can open another thread on this topic, if they really want to check it out.

closed 2016-08-17 by LongfeiProjects

davetcoleman, 2016-08-17:
@dk683 links work fine in comments, you just mixed up your brackets and parenthesis: [fixed url](https://gist.github.com/davetcoleman/4256083)