moveit/moveit_pr2 #86: Release into Kinetic?

130s, 2017-02-26:
We dropped Jade release https://github.com/ros-planning/moveit_pr2/issues/71
`pr2_moveit_config` is depended by https://github.com/ros-planning/moveit_tutorials/issues/65
Is there a plan to make a release into K? If so what has to be done?

I can take care of bloom stuff if desired.

v4hn, 2017-02-26:
> Is there a plan to make a release into K?

Yes.

> If so what has to be done?

Basically the same as for the jade issue. Just more.
Clearpath currently has their hands on the PR2 repositories but they do not support them officially anymore.
This will take some more time, but I'm confident we will see releases of the required ROS packages in kinetic.
It's on my ToDo list to deal with this.

k-okada, 2018-03-16:
pr2 on kinetic is now released, so please release mvoeit_pr2 into kinetic.

130s, 2018-03-19:
I don't have an extra gear for this any more. I appreciate if either @v4hn or @k-okada has an access rights and take this over.

v4hn, 2018-03-19:
It's on my todo list, but I'm not sure when I will get around to do that either.
Also I just noticed I do not yet have access to ros-gbp/moveit_pr2-release .

bmagyar, 2018-03-19:
If I'm given access to the required repos I can also help with releases from time to time.

130s, 2018-03-20:
@mikaelarguedas Sorry to pull you in, but assuming you have admin access can you add write access to @bmagyar @v4hn? Thanks in advance.

Maintainer field in package.xml is outofdate so @bmagyar @v4hn please consider adding yourself/ves there as well.

mikaelarguedas, 2018-03-20:
I don't have access to ros-planning ATM, maybe @v4hn or @davetcoleman have enough permissions to grant access to the source repository to the new maintainers? Otherwise I can look into getting the required permissions to add them.

I can give write access to the release repository. My understanding is that @bmagyar and @v4hn  will be added as contributors to the `moveit_pr2-release` repository and not added globally to the `ros-planning` team. Is that correct ?

130s, 2018-04-06:
@mikaelarguedas I think that's right. Thanks!

bmagyar, 2018-04-06:
Let us know once we have the permission and I'll go ahead with running some tests.

bmagyar, 2018-04-06:
I started testing some builds. It seems there are a few changes on rosdep keys and some C++11-related ones. I'll try to get to the end of it tomorrow and push a release to Kinetic.

Any plans for Melodic?

v4hn, 2018-04-07:
Please file pull-requests for any change you need to make in this repository before you release.

bmagyar, 2018-04-07:
Of course. All will be going against the `kinetic-devel` branch.

bmagyar, 2018-04-07:
WIP #96

davefeilseifer, 2019-03-30:
I've taken over maintainership of the pr2 packages and released almost all pr2 packages in kinetic. It might be worth some effort to try releasing the full suite of packages in kinetic again.

bmagyar, 2019-03-31:
This ticket should have been closed after I released it the last time but I can push a new one with the urdf compatibility changes, hold tight!

bmagyar, 2019-03-31:
I ran tests locally as the CI seems to be super slow today: https://travis-ci.org/ros-planning/moveit_pr2/jobs/513757640

PR:
https://github.com/ros/rosdistro/pull/20777

If you are happy, feel free to close this issue.

davefeilseifer, 2019-03-31:
I didn't open the issue, so I cannot close it

closed 2019-03-31 by rhaschke