moveit/moveit_pr2 #87: SIGABRT (std::bad_alloc)

ZdenekM, 2017-03-08:
When using pr2_moveit_sensor_manager/Pr2MoveItSensorManager, I'm gettig this:

```
...
[ INFO] [1488972345.413291806]: The cost of the trajectory is 0.769878, which is above the maximum safe cost of 0.010000. Attempt 2 (of at most 3) at looking around.
[ WARN] [1488972350.414088257]: Head moving action is done with state PREEMPTED: 
[ WARN] [1488972350.414169845]: Looking around seems to have failed. Attempting to recompute a motion plan anyway.
[New Thread 0x7fffb2572700 (LWP 5990)]
[ INFO] [1488972350.418757730]: Planner configuration 'left_arm[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1488972350.419652785]: left_arm[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[Thread 0x7fffb2572700 (LWP 5990) exited]
[ INFO] [1488972350.437487359]: left_arm[RRTConnectkConfigDefault]: Created 5 states (2 start + 3 goal)
[ INFO] [1488972350.437558891]: Solution found in 0.018639 seconds
[ INFO] [1488972350.439421806]: SimpleSetup: Path simplification took 0.001790 seconds and changed from 4 to 2 states
[ INFO] [1488972350.575553662]: The cost of the trajectory is 0.768883, which is above the maximum safe cost of 0.010000. Attempt 3 (of at most 3) at looking around.
[ INFO] [1488972350.858682712]: Sensor was succesfully actuated. Attempting to recompute a motion plan.
[New Thread 0x7fffb2572700 (LWP 6002)]
[ INFO] [1488972350.864043927]: Planner configuration 'left_arm[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1488972350.865438876]: left_arm[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[Thread 0x7fffb2572700 (LWP 6002) exited]
[ INFO] [1488972350.882566491]: left_arm[RRTConnectkConfigDefault]: Created 5 states (3 start + 2 goal)
[ INFO] [1488972350.882626636]: Solution found in 0.018348 seconds
[ INFO] [1488972350.885015599]: SimpleSetup: Path simplification took 0.002258 seconds and changed from 4 to 2 states
[ INFO] [1488972353.036064725]: Planning attempt 5 of at most 5
[New Thread 0x7fffb2572700 (LWP 6048)]
[ INFO] [1488972353.041141323]: Planner configuration 'left_arm[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1488972353.043620554]: left_arm[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1488972353.060479064]: left_arm[RRTConnectkConfigDefault]: Created 5 states (2 start + 3 goal)
[ INFO] [1488972353.060564317]: Solution found in 0.019024 seconds
[Thread 0x7fffb2572700 (LWP 6048) exited]
[ INFO] [1488972353.066723604]: SimpleSetup: Path simplification took 0.004564 seconds and changed from 4 to 2 states
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Program received signal SIGABRT, Aborted.
```

Moveit backtrace:

```
#0  0x00007ffff57b7cc9 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff57bb0d8 in __GI_abort () at abort.c:89
#2  0x00007ffff5dc6c9d in __gnu_cxx::__verbose_terminate_handler() ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff5dc4ce6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff5dc4d31 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff5dc4f48 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5dc544c in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff5e1d569 in std::string::_Rep::_S_create(unsigned long, unsigned long, std::allocator<char> const&)
    () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff5e1e34b in std::string::_Rep::_M_clone(std::allocator<char> const&, unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff5e1ea84 in std::string::assign(std::string const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007ffff3bd4468 in collision_detection::getCostMarkers(visualization_msgs::MarkerArray_<std::allocator<void> >&, std::string const&, std::set<collision_detection::CostSource, std::less<collision_detection::CostSource>, std::allocator<collision_detection::CostSource> >&, std_msgs::ColorRGBA_<std::allocator<void> > const&, ros::Duration const&) () from /opt/ros/indigo/lib/libmoveit_collision_detection.so
#11 0x00007ffff3bd4ba8 in collision_detection::getCostMarkers(visualization_msgs::MarkerArray_<std::allocator<void> >&, std::string const&, std::set<collision_detection::CostSource, std::less<collision_detection::CostSource>, std::allocator<collision_detection::CostSource> >&) () from /opt/ros/indigo/lib/libmoveit_collision_detection.so
#12 0x00007ffff506db5a in plan_execution::PlanWithSensing::computePlan(plan_execution::ExecutableMotionPlan&, boost::function<bool (plan_execution::ExecutableMotionPlan&)> const&, unsigned int, double) ()
   from /opt/ros/indigo/lib/libmoveit_plan_execution.so
#13 0x00007ffff508ee64 in plan_execution::PlanExecution::planAndExecuteHelper(plan_execution::ExecutableMotionPlan&, plan_execution::PlanExecution::Options const&) () from /opt/ros/indigo/lib/libmoveit_plan_execution.so
#14 0x00007ffff508fa4e in plan_execution::PlanExecution::planAndExecute(plan_execution::ExecutableMotionPlan&, moveit_msgs::PlanningScene_<std::allocator<void> > const&, plan_execution::PlanExecution::Options const&) ()
   from /opt/ros/indigo/lib/libmoveit_plan_execution.so
#15 0x00007fffc033dde8 in move_group::MoveGroupMoveAction::executeMoveCallback_PlanAndExecute(boost::shared_ptr<moveit_msgs::MoveGroupGoal_<std::allocator<void> > const> const&, moveit_msgs::MoveGroupResult_<std::allocator<void> >---Type <return> to continue, or q <return> to quit---
&) () from /opt/ros/indigo/lib/libmoveit_move_group_default_capabilities.so
#16 0x00007fffc033e986 in move_group::MoveGroupMoveAction::executeMoveCallback(boost::shared_ptr<moveit_msgs::MoveGroupGoal_<std::allocator<void> > const> const&) ()
   from /opt/ros/indigo/lib/libmoveit_move_group_default_capabilities.so
#17 0x00007fffc0350b63 in actionlib::SimpleActionServer<moveit_msgs::MoveGroupAction_<std::allocator<void> > >::executeLoop() () from /opt/ros/indigo/lib/libmoveit_move_group_default_capabilities.so
#18 0x00007ffff28d0a4a in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#19 0x00007ffff627d182 in start_thread (arg=0x7fffb3fff700) at pthread_create.c:312
#20 0x00007ffff587b47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

It does not happen if ```moveit_sensor_manager``` parameter is not set. Maybe important to note - I'm using two sensors to update octomap (PR2´s kinect + one external).

Any idea what might be wrong? Thanks!

v4hn, 2017-03-08:
This is weird indeed.

Could you try each sensor separately to see if it works with a single camera?
The alloc failure in the backtrace stems from the generation of cost markers.
This feature is (afaics) only active in debug mode and the string creation isn't memory intense, so I assume it's probably not the same bad_alloc with and without debug:=true.

Do you notice memory leaks in this part at runtime?

ZdenekM, 2017-03-08:
Ok, it seems that when I enable only one sensor (Kinect on PR2) it runs fine. I tried only for quite short period of time but without any crash. Do you have any tip how to check for memory leaks?

v4hn, 2017-03-08:
The other way around is just as interesting to test, probably even more so.
It could be that the manager doesn't deal with the other update-source correctly. Is that a Kinect too?

On Wed, Mar 08, 2017 at 06:58:50AM -0800, Zdeněk Materna wrote:
> Do you have any tip how to check for memory leaks?

How about opening htop on the machine running the move_group and looking at the memory usage? :)
Properly debugging leaks (e.g. with valgrind) can be quite painful with ROS projects...

-- 
Michael Görner, M.Sc. Cognitive Science, PhD Student
Universität Hamburg
Faculty of Mathematics, Informatics and Natural Sciences
Department of Informatics
Group Technical Aspects of Multimodal Systems

Vogt-Kölln-Straße 30
D-22527 Hamburg

Room: F-315
Phone: +49 40 42883-2432

ZdenekM, 2017-03-09:
Well, today I tried for longer period and it occasionally crashes even if there is only one sensor - kinect on PR2. It still crashes on ```getCostMarkers``` but the backtrace looks bit different so I'm posting it just for sure:

```
#0  0x00007ffff5e1ea4b in std::string::assign(std::string const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff3bd4468 in collision_detection::getCostMarkers(visualization_msgs::MarkerArray_<std::allocator<void> >&, std::string const&, std::set<collision_detection::CostSource, std::less<collision_detection::CostSource>, std::allocator<collision_detection::CostSource> >&, std_msgs::ColorRGBA_<std::allocator<void> > const&, ros::Duration const&) () from /opt/ros/indigo/lib/libmoveit_collision_detection.so
#2  0x00007ffff3bd4ba8 in collision_detection::getCostMarkers(visualization_msgs::MarkerArray_<std::allocator<void> >&, std::string const&, std::set<collision_detection::CostSource, std::less<collision_detection::CostSource>, std::allocator<collision_detection::CostSource> >&) ()
   from /opt/ros/indigo/lib/libmoveit_collision_detection.so
#3  0x00007ffff506db5a in plan_execution::PlanWithSensing::computePlan(plan_execution::ExecutableMotionPlan&, boost::function<bool (plan_execution::ExecutableMotionPlan&)> const&, unsigned int, double) ()
   from /opt/ros/indigo/lib/libmoveit_plan_execution.so
#4  0x00007ffff508ee64 in plan_execution::PlanExecution::planAndExecuteHelper(plan_execution::ExecutableMotionPlan&, plan_execution::PlanExecution::Options const&) () from /opt/ros/indigo/lib/libmoveit_plan_execution.so
#5  0x00007ffff508fa4e in plan_execution::PlanExecution::planAndExecute(plan_execution::ExecutableMotionPlan&, moveit_msgs::PlanningScene_<std::allocator<void> > const&, plan_execution::PlanExecution::Options const&) ()
   from /opt/ros/indigo/lib/libmoveit_plan_execution.so
#6  0x00007fffc006514e in move_group::MoveGroupPickPlaceAction::executePickupCallback_PlanAndExecute(boost::shared_ptr<moveit_msgs::PickupGoal_<std::allocator<void> > const> const&, moveit_msgs::PickupResult_<std::allocator<void> >&) () from /opt/ros/indigo/lib/libmoveit_move_group_pick_place_capability.so
#7  0x00007fffc0068bc0 in move_group::MoveGroupPickPlaceAction::executePickupCallback(boost::shared_ptr<moveit_msgs::PickupGoal_<std::allocator<void> > const> const&) ()
   from /opt/ros/indigo/lib/libmoveit_move_group_pick_place_capability.so
#8  0x00007fffc0080d30 in actionlib::SimpleActionServer<moveit_msgs::PickupAction_<std::allocator<void> > >::executeLoop() () from /opt/ros/indigo/lib/libmoveit_move_group_pick_place_capability.so
#9  0x00007ffff28d0a4a in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#10 0x00007ffff627d182 in start_thread (arg=0x7fffb3574700) at pthread_create.c:312
---Type <return> to continue, or q <return> to quit---
#11 0x00007ffff587b47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

ZdenekM, 2017-03-09:
Btw, this part of code https://github.com/ros-planning/moveit_pr2/blob/indigo-devel/pr2_moveit_plugins/pr2_moveit_sensor_manager/src/pr2_moveit_sensor_manager.cpp#L65 looks a bit suspicious to me - how is it supposed to work with multiple sensors?

v4hn, 2017-03-24:
Ok, I finally had a moment to look at this...
Your problem is most likely not at all related to multiple cameras.
The sensor manager concept is a straight forward version of the replanning concept:
If the planned path exceeds a threshold in terms of potential collisions, it might help to point a sensor at the most "risky" location to improve the environment model (mostly by getting an updated octomap in that region).

`Pr2MoveItSensorManager` implements this "pointing" functionality for the head, using the PR2's `PointHeadAction`. If you want to implement this for another orientable camera, *then* you have to model multiple cameras here. Otherwise the plugin should work as is, also with multiple cameras to update the octomap.

If you didn't do that already, could you check how many elements are in `cost_sources` above [this line](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/plan_execution/src/plan_with_sensing.cpp#L195)?
I suppose you work with a source build?
Also I would be interested in whether you also get bad_allocs if you just comment out this block.
It's just publishing markers anyway.

rhaschke, 2019-07-04:
Closing due to now activity for >2 years.

closed 2019-07-04 by rhaschke