moveit/moveit_pr2 #93: The kinematics plugin (right_arm) failed to load.

maxgittelman, 2017-11-30:
Hello,

I am working though the [MoveIt! RViz Plugin Tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ros_visualization/visualization_tutorial.html) and noticed an error in the launching of the demo file.

When executing: `roslaunch pr2_moveit_config demo.launch`, I noticed the following Error:

[image]

Why should I be or not be worried by this?

davetcoleman, 2017-12-01:
Is this on Kinetic? The PR2 has poor support on kinetic

maxgittelman, 2017-12-01:
Yes, this is on Kinetic

rhaschke, 2018-02-27:
Try to use KDL IK plugin instead.

davetcoleman, 2018-03-26:
I believe @rhaschke provided a good solution to this

closed 2018-03-26 by davetcoleman