moveit/moveit_pr2 #104: Release versions messed up?

rhaschke, 2019-07-04:
Dear Isaac,
is there a special reason that you released `moveit_pr2` into Indigo as 0.7.15, although the 0.7.x is reserved for Kinetic and the next Indigo release should have been 0.6.7?

[image]

v4hn, 2019-07-05:
Four days before the `0.7.15` release here, moveit in indigo was released as `0.7.14`.
I guess this is where the wrong version number comes from.

To resolve the crossed versions and synchronize this repository with the main moveit versions again, I would propose to jump ahead and update the `kinetic` version to 0.9.x (0?) with the next kinetic release.

Does that sound ok to you @rhaschke ?

rhaschke, 2019-07-05:
Yes, that would sync with version numbers of ROS. Fine with me.

130s, 2019-07-06:
TLDR; I admit my mistake. I agree with bumping kinetic to 0.9.x

Although I can't find a PR for the change in question, I see commit message https://github.com/ros-planning/moveit_pr2/commit/c09bca016067da7a0e8eb95ff02b92781848db06 saying:

>  Package version aligning with a newly added moveit_full_pr2. This is needed for the release into rosdistro

That's where `moveit_full_pr2` pkg was moved into this repo. That package was in 0.7.x already before merge. I wanted to stick to using release util tools `catkin_prepare_release`, which allows us more organized systematic release handling but does require all packages in a repo to be the same version. Choice was to either downgrade the new pkg to align with 0.6.x or to upgrade the existing pkgs to 0.7.x. Since forcefully downgrading version is not a good idea in general (it would cause another problem e.g. potentially colliding with existing version), I chose upgrade. Up to here I hope it isn't that bad.

I think my fault was that I didn't look around to see if there was already 0.7.x release back then. Without doing so, I bumped minor version from 0.6 to 0.7 for all the existing packages in this repo. Apparently there was already [0.7.0](https://github.com/ros-planning/moveit_pr2/releases/tag/0.7.0) earlier that year.

rhaschke, 2019-07-07:
Thanks for the explanation. Let's release as 0.9.x next.

closed 2019-07-07 by rhaschke