moveit/moveit_pr2 #106: Request for release on melodic

k-okada, 2019-10-01:
@rhaschke could you release new version including  https://github.com/ros-planning/moveit_pr2/pull/101 and run `blooms into melodic? Or if you’re busy please consider set this repository as orphaned as other pr2 repository (https://github.com/PR2/pr2_common) so that @v4hn and @k-okada can release this?

rhaschke, 2019-10-01:
I am not the main maintainer of this package, but I prepared a release for melodic as requested:
https://github.com/ros/rosdistro/pull/22478

I think @v4hn should be able to release himself.

k-okada, 2019-10-01:
Great! Thank you!!!

v4hn, 2019-10-07:
Thank you for the release Robert.

You are right, I am probably in charge of this package by now, although somewhat against my will.
As we have to use custom versions of these packages for the modified PR2 in our lab, I do not work with this repository at all.

bmagyar, 2019-10-08:
Hi guys,

FYI all melodic builds are still failing, how come this wasn't picked up the pre-release test?
http://build.ros.org/job/Mbin_ubv8_uBv8__pr2_moveit_plugins__ubuntu_bionic_arm64__binary/

If you want to be the maintainer, please replace me in the package.xml but I can also keep releasing `moveit_pr2` although I don't have a robot to test it on.

rhaschke, 2019-10-08:
Hi @bmagyar, I have pushed a fix and requested a new release: https://github.com/ros/rosdistro/pull/22585 a few days ago.
However, this release PR is not yet merged. So, the build farm will continue to report errors...
I'm not interested in maintaining this package at all. I was just asked to do a release, which I did as a MoveIt maintainer.

rhaschke, 2019-10-08:
Release PR was merged and ros build farm is happy now:
http://build.ros.org/view/Mbin_uB64/job/Mbin_uB64__moveit_pr2__ubuntu_bionic_amd64__binary.

closed 2019-10-08 by rhaschke